limikael/freecad-slic3r-tools #1: Is this Workbench py3/qt5 compatible ?

luzpaz, 2019-05-13:
What versions of FC does this WB support? FC has been migrating to py3/qt5, does this workbench support py3/qt5 as well ? Does it also support py2/qt4? 

BTW, nice job

luzpaz, 2019-05-13:
Also are there any 3rd party dependencies needed?  
What is the minimum version of slic3r?  
Minimum version of FC?  
Any known issue?

limikael, 2019-05-15:
It is py3/qt5 compatible. However, I don't think it is py2/qt4, mainly because of laziness on my part. I will do some more testing and provide documentation of the results.

luzpaz, 2019-05-15:
I'm not saying you should make it py2/qt4 compatible. Was just wondering.

limikael, 2019-05-16:
Ah ok... I also asked here:

https://forum.freecadweb.org/viewtopic.php?f=10&t=36398

limikael, 2019-05-16:
Updated the docs

closed 2019-05-16 by limikael