limikael/freecad-slic3r-tools #5: Mention a dedicated thread on the FreeCAD forum for people to give you feedback about this workbench

luzpaz, 2019-05-13:
Make it separate from announcement thread that you created. Then you can cross-link the URL to said thread in the README.md

limikael, 2019-05-15:
Good idea! Where should I make this thread you think? Can you point me to a similar thread for another workbench so I can see the format for the thread?

luzpaz, 2019-05-15:
Examples: 
* https://github.com/shaise/FreeCAD_FastenersWB/tree/master#feedback  
* https://github.com/triplus/Glass/tree/master#discussion  
* https://github.com/amrit3701/FreeCAD-Reinforcement/tree/master#extras

luzpaz, 2019-05-16:
Nice. Not sure what commit you added this in but I'll close this ticket nevertheless

limikael, 2019-05-17:
Oh... I haven't actually made the thread yet...

limikael, 2019-05-17:
Ah I see... I did make a link to my old thread at the same time as I asked about the examples for the formats... Will create a new thread and change the link.

limikael, 2019-05-17:
On 2nd thought... I think I'll use the thread that is already there, since it already has some discussions in it... so closing again... :)

closed 2019-05-17 by limikael